happycube/ld-decode #10: RF Capture quality metrics

simoninns, 2018-10-01:
In order to judge the quality of RF captures (in terms of disc and capture-player quality) it would be good to have a 'bogomips' style metric based on a number of technical factors which provides a single 0.00 to 9.99 style grading of a capture sample.  Once many RF images are available (especially of the same disc) this will become a valuable first 'sort' to quickly find the best source RF captures.

I propose a scale of 0 to 10 happycubes as the measurement unit :)

simoninns, 2018-11-16:
Once the JSON metadata is included in ld-decode this can be solved using the VITS object to output information about the SNR (and other VITS related analysis). This can be done within ld-decode or by a video post-process tool.

happycube, 2019-01-27:
Planning on adding SNR and line 19 chroma level (which seems slightly higher than burst on NTSC _outside the low-MTF area_, which means there might be some misalignment) before rev4 merge.

happycube, 2019-01-29:
part 1 complete at least

closed 2019-01-29 by happycube