happycube/ld-decode #8: Signal analysis test-bench

simoninns, 2018-10-01:
It would be nice to create a test-bench for the ld-decode process with a known set of input signals for each stage.  As development ramps up it will be important to be able to rapid test new code for signal processing quality as well as regression testing with something more exact than a real LaserDisc sample (possibly software-generated test images).

happycube, 2018-10-06:
A test bench would be very good indeed... I'll also accumulate a set of <=10 frame test samples from real disks that have been problematic (i.e. the WotW sample with a destroyed line) 

I wish I had bought that LD RF generator that was on Y!J for a long time, but it would have been quite expensive.

simoninns, 2018-10-07:
It should (as an additional thought) be possible to code a simple signal generator that makes a series of test RF images (generate video and audio - modulate and 'simulate' the output from LaserDisc).  I realise it's not a super-simple idea, but worth noting here.

happycube, 2019-01-27:
Temporarily moving this up, since I'll start a sample library after rev4 merge.

happycube, 2019-01-29:
*holds out empty coffee can with dice, shakes*

Anyone got specific difficult/troublesome samples 1-2 seconds long for the collection?

happycube, 2019-01-30:
Working on it :)  I have some NTSC comb filter centric samples at the moment.

happycube, 2019-02-02:
Moving this to rev5, it's making progress but isn't finished yet.  Need some audio samples for CX, for instance...

happycube, 2019-05-19:
Moved to rev6 since this is ongoing

simoninns, 2019-11-11:
Moving to rev7

simoninns, 2019-12-22:
Closed (replaced with new issue about using GitHub CI rather than manual tools).

closed 2019-12-22 by simoninns