happycube/ld-decode #13: Make a Wiki on Github

simoninns, 2018-10-02:
Let's start on documenting this awesome piece of code!  Time to make a Wiki :)

simoninns, 2018-10-02:
Add in  the initial install instructions, something like:

sudo apt install clang
sudo apt install libopencv-dev
sudo apt install libfann-dev
git clone https://github.com/happycube/ld-decode
make

sudo apt install python3-pip
pip3 install numpy
pip3 install pandas
pip3 install scipy
pip3 install matplotlib
sudo apt install python3-tk 

python3 lddecode.py

simoninns, 2018-10-02:
More instructions for the current commit:

Decode:
python3 lddecode.py input.lds output 0 25

0 25 means - start at position 0 and continue for 25 frames

Note: if it's a whole disc you need to miss the lead-in with something like 1000 25 (start at 1000 and continue for 25 frames)

NTSC
display -size 844x505 -depth 16 gray:output.tbc

PAL
display -size 1052x610 -depth 16 gray:pal2.tbc

NTSC Comb filter:
cat output.tbc | ./comb-ntsc -d 2 > output.rgb

Display RGB NTSC
display -size 744x480 -depth 16 rgb:output.rgb

simoninns, 2018-10-02:
Update :)

NTSC
display -size 844x505 -depth 16 gray:output.tbc

PAL
display -size 1052x610 -depth 16 gray:pal2.tbc

NTSC Comb filter:
cat output.tbc | ./comb-ntsc -d 2 > output.rgb

PAL Comb filter
cat output.tbc | ./comb-pal -d 2 > output.rgb

Display RGB NTSC
display -size 744x480 -depth 16 rgb:output.rgb

Display RGB PAL
display -size 974x576 -depth 16 rgb:output.rgb

Convert is like display but converts the .rgb into another format (PAL example):
convert -size 974x576 -depth 16 rgb:output.rgb ~/rfcapture_output/frame$i.png

Extra:
use the -m option to get a live view
(in comb-*)

happycube, 2018-10-06:
I think this can be closed now, since there is a wiki :)

closed 2018-10-06 by happycube