happycube/ld-decode #14: lddecode.py crash during long decode process

simoninns, 2018-10-04:
After processing 42,728 frames of NTSC CAV RF capture the lddecode.py command crashed with the following output:

1601
starting at  70935161684
status code 9150521
True None
starting at  70935827859
status code 9150521
False None
frame  42727
1603
starting at  70936496358
status code 9150521
True None
starting at  70937162533
status code 9150521
False None
frame  42728
1601
starting at  70937831028
status code 9150521
Traceback (most recent call last):
  File "lddecode.py", line 39, in <module>
    combined, audio, nextsample = framer_ntsc.readframe(fd, nextsample, f == 0)
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 1027, in readframe
    f, sample = self.readfield(infile, sample)
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 992, in readfield
    f = self.FieldClass(self.rf, rawdecode, 0, audio_offset = self.audio_offset)
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 958, in __init__
    self.linelocs3, self.burstlevel = self.refine_linelocs_burst(self.linelocs2)
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 845, in refine_linelocs_burst
    scaledburst, audio = self.downscale(outwidth=self.outlinelen, channel='demod_burst')
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 921, in downscale
    dsout, dsaudio = super(FieldNTSC, self).downscale(audio = final, *args, **kwargs)
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 615, in downscale
    scaled = scale(self.data[0][channel], lineinfo[l], lineinfo[l + 1], outwidth)
IndexError: list index out of range

The command line was as follows:

python3 lddecode.py ~/Capture/War\ of\ the\ worlds_CAV_NTSC_side1_2018-10-03_06-24-34.lds ~/Capture/wowoutputntsc 0 60000

This could be related to the end frame specified as 60,000 (which is more than the frames present in the capture file), however the last frame of the capture should be 43,132 according to the original disc.

Test was performed against commit: d69c3b10ed44d8bfe46d6905631975df5e9e09da

happycube, 2018-10-05:
I believe this is fixed with the latest version.

simoninns, 2018-10-07:
I'm not going to have time to do a full decode to completely test this, so I'll close the issue and reopen it later if the problem reoccurs.

closed 2018-10-07 by simoninns