happycube/ld-decode #15: PAL Analogue audio decode incorrect

simoninns, 2018-10-05:
Steps to recreate:

python3 lddecode.py /SFTP/LDV4300D_2\ Captures/AIV/Domesday_DiscSet2/Domesday\ Community\ North_CAV_PAL_side1_Disc\ set\ 2_2018-10-02_15-11-01.lds ~/Capture/output_dd_cn 600 500

Command produces tbc and pcm files.  PCM file created is attached; sound contains both original signal and a constant buzzing.

[output_dd_cn_audio.zip](https://github.com/happycube/ld-decode/files/2449814/output_dd_cn_audio.zip)

simoninns, 2018-10-06:
Testing with commit 0484282 show significant improvement; the constant buzz is now gone.  There is still quite a lot of clicking in the file though.  Sample (created with the same command as above) attached.

[output_dd_cn_audio.zip](https://github.com/happycube/ld-decode/files/2452782/output_dd_cn_audio.zip)

happycube, 2018-10-06:
Some of the remaining issues are from disk dropouts - compensation hasn't been implemented for audio or video yet - and others may be mistuning.  NTSC analog audio seems to be working correctly when looking at a 1khz wave, but until PAL reference/service disk captures are available it will be hard for me to debug PAL analogue audio properly.

happycube, 2019-11-18:
Tracked down the severe popping to a TBC issue, still investigating and will hopefully fix soon.

happycube, 2019-12-01:
This should be getting a lot better, a round of sanity checks should be done before closing this though and confirmation that the hz is actually correct once #361 and #293 are finished.

happycube, 2019-12-22:
"

closed 2019-12-22 by happycube