happycube/ld-decode #16: Wanted: VHS RF captures

happycube, 2018-10-06:
At some point (after LD is settled) I'd like to also decode VHS RF.  If someone with a DdD setup can get VHS captures, this might get sped up a bit :)

vonj, 2019-02-02:
> I'm interested in working on VHS decoding. Haven't got a DdD yet, though my test capture (PAL) with cxadc already gave some interesting results. I'm planning to get my hands on a DdD though.
>
> Using the old vhs-decoder.py script in the repo on the capture gave me this, after adding some gain in the post-demodulation stage:
>
>
>
> So it seems I tapped the right test point at least.
>
> Not sure how to best organize it code wise though. Got to implement colour decoding, and maybe make it output something compatible with the current tools like tbc. Also need some way of synchronizing the video to audio.
>
> Betamax and the video portion on 8mm video is very similar to VHS so it may not be a huge step from VHS to decode those as well.

Oh my god. This is the holy grail!

happycube, 2019-02-03:
I'd say a grail-shaped beacon, myself ;)

Seriously, this is very cool.  Dunno when I'll have time to dig into this, but it should be possible to adapt the RF code in current ld-decode to deal with VHS.  It'd be handy to have a Jupyter notebook for research.

In this sample I'm seeing that deemphasis isn't right yet.

oyvindln, 2019-02-03:
I would be happy to help out on the coding side of it. I'll have to look into jupyter, and see if I can start making some notes.

VHS uses "color under" decoding, so the chroma is stored separately from the luminance on the tape, and in the rf signal at a lower frequency (and is not fm modulated like the luma as far as I've read?). So you would have to decode and store that separately, rather than decode it from the intermediary demodulated rf like is done in the laserdisc toolchain currently. 

https://www.av-iq.com/avcat/images/documents/pdfs/tt189%20-%204611.pdf

vonj, 2019-02-04:
Would you need different hardware?

oyvindln, 2019-02-08:
Ok, result after first hacky attempt to decode with ld-decode.py with some small tweaks:

[image]

Still need a lot of tweaking of course, but this looks very promising already

> Would you need different hardware?
> […](#)

Not sure what you mean? 

For this test I captured with cxadc, which is a modified driver for capture cards based on the Conexant cx2388x video decoder chip. The driver puts it into some raw mode that makes it output the raw signal (at 28.6 MHz sampling frequency, with 8bit resolution.) The DdD which is used for the laserdisc captures now is 40MHZ 10-bit so it may give a nicer signal, but I don't know as I haven't got one yet.

I used a standard BNC test probe (probes that came with my analog discovery 2 usb scope), using a BNC to RCA adapter to connect it to the capture card. The probe was connected the rf test point on a SONY SLV-SE60N VHS deck, a mid-range 4 head Hi-Fi deck. So nothing fancy. (Had to use the s-video connector on the capture card for whatever reason, so used a S-Video-> RCA dongle from a graphics card I had lying around in reverse to connect.)

Had to adjust the IRE zero-point in the decoder to pick up the syncs correctly. I just eyeballed it from a signal plot, not sure how to calculate it properly, and also disable some audio stuff in the code to avoid it crashing.

Will look at it some more tomorrow.

oyvindln, 2019-02-18:
Did some more color bar captures and capture from a commercial VHS release, even managed to decode a S-VHS ET (S-VHS on normal VHS tape) section.

Haven't had a lot of time since the last update to tinker with this otherwise.

(Of course I didn't  realize that the input level in cxadc can be adjusted before after doing this.)

For integrating VHS (and other non-LD sources) into the decoder, I think it would make sense to define some input "types" like the already defined [sysparams](https://github.com/happycube/ld-decode/blob/9f6b9207a90330f4207e06e99bc1e4b1a8865d0b/lddecode_core.py#L28) structures. Maybe even make them loadable from a JSON file.

In a VCR, the color subcarrier is upconverted to the normal PAL/NTSC frequency when decoding. In normal VHS decks it's mixed in with the demodulated luma signal for the composite output, or in S-VHS decks, output separately with S-Video. It's possible one could skip the upconverting step in the software decoder and decode the color directly from the lower frequency, but I don't know for sure.

Ideally one would want to avoid the Y/C mixing before doing colour decoding to avoid introducing crosstalk issues, though maybe it is easiest to start out with upconverting and mixing as that would allow using the existing comb filter without any modification. Later I think it could be output as a separate .tbc (or similar) file and mixed after the colour decoding is done.

I presume the current tbc code also looks for color bursts at the normal frequency, maybe that's why the current output is a bit wiggly and not fully corrected.

Additionally, for decoding the VHS I did have to disable some audio stuff as the code assumed there would be audio, and specifying the sampling frequency as a non-integer (needed for correct cxadc frequency settings) gave an error so I had to work around it. Those things should be simple to fix/add settings for.

atsampson, 2019-02-18:
Look at all that nice Teletext data in the capture as well - it'd be nice to be able to hook this up to [vhs-teletext](https://github.com/ali1234/vhs-teletext/)'s decoder...

oyvindln, 2019-02-18:
Yeah, there [are already](https://www.facebook.com/teletextarchaeologist/photos/a.1458600467574521/1725210967580135/?type=3&theater) some decoded captures from ld-decode of the teletext data on the "British Garden Birds" laserdisc, though some proper integration would be nice.

ivitaly, 2019-02-25:
I can help with writing some cpp/h files like wrapping some math formules used in BT601 REC709 etc reccomendations if needed, like port something or wrap in C/C++/C# if there is some need in that languages in project. 

The RF (raw sample of tapes) It's like reinventing the WAV format, when there is only mp3.
I guess that in 2019 computer can process video/audio streams at ultimate quality and formats should be rethinked at all (except considering that we code information somehow wrong, and minutes will take terabytes of data)
I guess we can made our own loseless video format, something better than h265, like modern digital cinema cameras RAW to do superior processing when editing.

If I can help anyhow with my 15+ years of C, feel free to contact me : IsVitaly@gmail.com

wikt0r, 2019-02-25:
> I'm interested in working on VHS decoding. Haven't got a DdD yet, though my test capture (PAL) with cxadc already gave some interesting results. I'm planning to get my hands on a DdD though.

I would like to ask you about the VCR you used. What type is it? Should you modify the VCR to capture the RF signal or it has default output for it? Thank you!

oyvindln, 2019-02-25:
The VCR I used for this test capture was a Sony SLV-SE60. It's a pretty standard 4 head hi-fi VHS deck. From what I can find it's from around 1999.

I did not modify the VCR in any way. VCRs typically have a test point with the rf output from the video heads which is used to calibrate tracking and the tape guides that hold the tape around the video head drum. This is what I connected to to get the raw rf signal.

The main issue I had was that the rf output level was pretty low. It's possible to adjust the input level a bit on the capture card, it was set at 16 by default (and on the capture) but can be turned up to 31, which I will try next time. The test points seem to normally output the rf signal before it's amplified in the player, so maybe it would be possible to tap it later in the chain, though that may be a bit more difficult to access.

Yesterday I also tried to capture from a JVC HR-J658E VCR, but with this one the rf output was too low, so even with the max input level on the card it was much lower than that of the sony and too low to decode properly. For comparison, when capturing the composite output, the input level had to be turned down significantly to not clip.

Capturing with the DdD board would probably be better, I've ordered one now but it may take some time to get it set up.

happycube, 2019-03-04:
As for your IRC question about how to integrate it - it'd be good to have a data structure that describes the medium in question, which would then be adjusted by command line parameter as needed.  What exactly that *is* I/someone hasn't worked out yet, but right now what little spare head-capacity I have is going to EFM.

I think the end result will be more elegant all around, it'll just have to wait for rev6 :)

ivitaly, 2019-03-08:
I found some neural network paper, maybe use it as pre pass filter in cuda to improve vhs captured signal?

https://www.researchgate.net/publication/3183174_Cascaded_Neural_Network-based_S-VHS_Restoration

vonj, 2019-03-08:
If I am not mistaken, such things are better applied *after* ld-decode is done. So more of a post filter rather than pre pass.

I am thinking more like a step in an AviSynth filter chain.

oyvindln, 2019-03-10:
As suggested on IRC, I have a fork set up [here](https://github.com/oyvindln/ld-decode). For now it has the bare basics to do black and white PAL VHS rf decodes.

The VHS-specific stuff is separate from the original ld-decode code for now, using simple inheritance.

ivitaly, 2019-03-27:
Curious someone think about to make board with quantizing chip like usb whistle that have input as s-video, output as usb type c?
I guess best results is to change HW firmware to work with signals Y/C properly like YC->LD RF converter

simoninns, 2019-12-22:
Since this isn't really an ld-decode issue, I'm closing it.  This is not to say I don't love the conversation, but I'm guessing there is a VHS repo now to continue it on :)

closed 2019-12-22 by simoninns