happycube/ld-decode #17: PAL output levels clipping white

simoninns, 2018-10-07:
Here is some feedback from the work on the PAL comb filter (copied from an email message):

The captures are too "hot", and the scaling is clipping the chroma peaks in the composite signal.

Attached image shows the "oscilloscope" trace of a line in the colourbars near the top of your TestCard F capture (From Domesday CommN). The height of the white rectangle represents the range 0-255 in 8-bit code. You can clearly see the top of the chroma is clipped.

(Richard Russell said somewhere on one of his testcard pages that Test Card F has "95%" colour bars)
The clipping is worse on the colour-bar images later in that sequence with the red bottom-half of the screen, that are probably 100% colour bars.

For background, see also the images about 2/3rds of the way down this page: https://www.radios-tv.co.uk/Pembers/World-TV-Standards/index.html
and figure 2 on this page: https://www.embedded.com/print/4009973

More specifically, see https://tech.ebu.ch/docs/tech/tech3280.pdf  for advice/standards on levels for digitised composite video (there may also be something in Rec.470 or Rec.601 but I haven't looked recently)

For digitised composite (PAL), in 8-bit:

- sync-tip should be at code 01h
- blanking level should be at code 40h  (64 decimal)
- peak-white should be at code D3h (211 decimal)

This will give sufficient headroom for the chroma on, for example, the yellow bar of the chroma bar, to go above peak-white, as it needs to.

In your captures of testcard F, for example, blanking level is at about 40 (decimal) and peak white at around 240 (decimal).

I can see that the chroma on the testcard F is clipping - which will reduce the decoded saturation and luminance, of the yellow and, because the chroma is clipped so not sinusoidal, it'll tend to leave 2nd and 3rd harmonic chroma pattern remnants in the image after decoding.

(This is why I'm having to set my video-gain and colour-saturation parameters to 75% rather than 100%, to get acceptable results.)

It'd be good to get the parameters tweaked in the RF demodulation part of the ld-decode software to remedy this (As you're doing 16-bit codes, you won't lose anything in granularity)

It's not the end of the world, but the present result is definitely sub-optimal.

(The recommendations for digitised NTSC are probably similar, though not exactly the same)

[image]

happycube, 2018-10-07:
There's a good chance that I need to apply MTF compensation on top of other adjustments - basically there's a lack of bandwidth on CLV laserdiscs and the first few thousand frames of CAV laserdisc, and I need to add an automated method of applying the filter for that.  In addition I can adjust the output levels for PAL to match 16-bit versions of those values soon.

techmindorg, 2018-10-07:
I've just checked a new capture by Simon, with the fixed code, and the levels look good now.
I wouldn't worry about the MTF - I haven't measured it technically, but now the levels have been fixed the colour saturation eyeballs well.

Bear in mind that the test sequence (that Simon has sent me, output_dd_cn_fix.tbc ), contains at least three different levels of colourbar. I believe the Testcard F colour bar at the top is "95%" saturation, the full screen colours bars later in the sequence are "75%" and the few frames which are only half a frame of colourbars with a red bottom-half of image are probably 100% bars.

My program was mathematically followed through (for gain) at one point, and was (at one point) checked with a digital composite image generated by Richard Russell, so as the image saturation now looks good when I set my "video gain" and "chroma gain" both to 100% - I wouldn't be any any hurry to change anything - especially if it means adding complexity. If you have specific evidence for why you think it's wrong, then let's discuss!

See image below, "scope trace" is from colourbars at top.

[image]

techmindorg, 2018-10-07:
It is possible that my program self-calibrates the chroma level based on the burst-amplitude, so the colour-saturation of the decoded image may be tolerant (immune) to moderate high-frequency gain errors in the source material. :-)  This is something I experimented with in the past, but I'm not sure if that feature is active in the present code!

techmindorg, 2018-10-12:
By the way, I've just been checking the definitive PAL reference, Rec.470, https://www.itu.int/dms_pubrec/itu-r/rec/bt/R-REC-BT.470-6-199811-S!!PDF-E.pdf

According to page 13 of that, the peak-to-peak amplitude of the subcarrier burst should be "3/7 of difference between blanking level and peak white-level ±10%.  For systems D and I the tolerance is ± 3%". (UK standard is PAL I). So if video is sampled so black = 64, white = 211 (as previously discussed), the peak-to-peak amplitude of the subcarrier burst should be nominally 63 digital counts. Owing to digital sampling effects, ideally the true amplitude would be determined by "fitting" a sinewave or demodulating it by product-detection.