happycube/ld-decode #18: PAL decode - misalignment of video lines

simoninns, 2018-10-07:
There is some form of misalignment of the video lines in PAL decodes.  Please see the attached TBC output:

[image]

techmindorg, 2018-10-07:
happycube, we think this issue may have been introduced fairly recently - still-frame captures that Simon got from the Doomsday disc a while back don't have the horizontal field misalignment.

Are you using phase information from the chroma burst to fine-tune the TBC timing? (This is definitely a good idea in principle)
The chroma burst pattern looks "wrong" in the misaligned captures; when fields have been interleaved the chroma burst it should be a *fine* dot pattern. It may be that your program isn't anticipating the intricacies of the PAL chroma burst with its V-switch :-)  . Let me attach a couple of images to explain...

First image here is a crop from testcard F, this shows the WRONG burst alignment between fields.
[image]

Image below is a crop from the still-frames sequence Simon captured (with older code?) a while back, which shows correct (or much nearer correct) field-interleaved burst pattern.
[image]

I don't suspect they're doing anything special for LaserDisc. Do you have a copy of Rec.470 ? All the details will be in there.

techmindorg, 2018-10-07:
(By the way, this won't be affecting my comb-filter/colour-decode as I'm treating the two fields completely independently.)

simoninns, 2018-10-07:
There is a 'special' thing here just for PAL LaserDiscs called the video signal pilot burst; it's an additional synchronising pulse of 3.75MHz contained within the sync.  This is covered by IEC 60856-1986; here's the diagram of it from the specification:

[image]

From what @happycube mentioned to be earlier, it seems there is some issue with the code that locks on to that.  Nothing more to add really, but I thought it good to share the diagram with you @techmindorg

techmindorg, 2018-10-07:
Cool!

simoninns, 2018-10-13:
This looks fixed in the current commit - closing

closed 2018-10-13 by simoninns