happycube/ld-decode #19: Colour banding in PAL decodes

simoninns, 2018-10-11:
In several PAL decodes there seems to be a form of colour banding.  The following image is post-colourization, but the banding is visible in the TBC output:

[image]

decode command was:

python3 lddecode.py /media/sdi/Seagate\ Backup\ Plus\ Drive/SFTP/Jason\ ATA\ Disc\ 4/JasonPAL_d4_tc1.lds ~/Capture/hc_test3 0 10

Frame #4 (although it shows on the other frames too)

Commit: b93ef4e7991100dfee0bdaa696ed44c43eba016b

techmindorg, 2018-10-12:
I suspect this may relate to subpixel timing jitter. If the banding is only visible on saturated colours and disappears on neutral colours (whte/greys) that pretty much confirms it. (You may need to turn down the "brightness" a little to see the white without risk of saturating.)

It'd also be worth trying testing the PAL filter to not do interline filtering...
In code https://github.com/simoninns/ld-colourfilter/blob/master/palcolour.cpp
In lines 284 to 302, try making yfilt2 =0 (first) then also cfilt1,2,3  =0. (leaving just yfilt0 and cfilt0 as-is)
This will give a softer image, more cross-colour, and may reveal some alternate-line hue-shifts (so not a good long-term work-around) but if this eliminates or markedly reduces the horizontal banding, then again this would tend to implicate TBC sub-pixel jitter.

Of course it could be something else in the RF demodulation...

simoninns, 2018-12-24:
I'm going to close this; it's pretty much covered by issue #73 now.

closed 2018-12-24 by simoninns