happycube/ld-decode #31: Auto detect PAL or NTSC

simoninns, 2018-10-18:
As a future enhancement, it would be great if the decoder could automatically work out the input video standard.

happycube, 2019-08-19:
Pushing to rev7, since it's a little bit involved - basically it needs to detect that it's seeing the other standard, then the program will all but restart with the other standard's settings.

atsampson, 2022-05-17:
A lower-tech fix would be to take advantage of DomesdayDuplicator's filename convention: enable `--pal` if the input file's basename contains the string `_PAL_`.

(Similarly, `_ANA_` could imply `--noEFM`, doing the right thing for PAL/NTSC audio tracks by default - I must submit a DDD patch to add a checkbox for that at some point...)

closed 2025-12-09 by simoninns