happycube/ld-decode #34: Field ordering does not seem harmonised

simoninns, 2018-10-18:
When dealing with NTSC CAV, NTSC CLV, PAL CAV and PAL CLV the field order/alignment doesn't seem to be the same in the output from ld-decode.

Since downstream applications cannot determine the field order easily (and the output is always interlaced), ld-decode should output all disc and video types in the same way otherwise things like VBI decoding will be very difficult to implement.

techmindorg, 2018-10-19:
I think I read somewhere that field ordering has not always been consistent (e.g. when telecine of 24 or 25fps film to video), especially not in the 1970's.
I'm curious as to why the design decision was made for ld-decode to output the fields interleaved from the TBC rather than in a field-sequential more native CVBS format...
PAL has an 8-field sequence of burst and V-switch phase relative to the line-syncs, so (as long as the edges of the line-syncs are still present in the output files) in principle the field-ordering can be determined later...  but you wouldn't want to do this if it could be avoided!

simoninns, 2018-11-16:
This is solved by the --field option which no longer required ld-decode to understand the frame order

closed 2018-11-16 by simoninns