happycube/ld-decode #38: ralf still off at top/bottom

happycube, 2018-10-22:
Probably a small bit of 1D fallback at top/bottom.

https://diff.pics/US7Tl7teXb-v/1

Gamnn, 2018-10-22:
It _might_ only be at the top. If you're looking at the diff picture, the white line near the bottom and rectangle shows where it's improved, the first two lines at the top is where it's regressed.

It's a little hard to tell on Ralf since the bottom is a little busy. The shot of the kids at the start of the RF _might_ indicate the bottom line is a little off though. https://i.imgur.com/TwWwerg.png

happycube, 2018-10-24:
Should be better now.  I saw that the 3D comb filter wasn't working correctly at the bottom, falling back into 2D mode.

happycube, 2018-11-06:
After a regression I fixed this morning it should actually be OK now. :)

closed 2018-11-06 by happycube