happycube/ld-decode #40: Seeking for chapters

Gamnn, 2018-10-31:
Being able to TBC from and to chapters would be helpful

Perhaps:
--from-chapter, -f
--to-chapter, -t

So _-f 3 -t 3_ would process all of chapter 3. _-f 1 -t 5_ would do chapters 1-5.

happycube, 2018-12-24:
This is tricky since there's no way to know where chapters begin or end and it takes a long time to do the actual seeking right now. 

I wouldn't expect a fix for a while - it's probably better to let the player do it, and then bug me about adding CLV timing as a parameter for -S ;)

happycube, 2018-12-24:
Setting this to rev5, since it _might_ become viable if we get TOC data.  If not, this probably won't happen.

happycube, 2019-08-19:
Chapter seeking might be more reasonable now that ld-decode is faster... basically it needs to start at the beginning and jump ~10 seconds or so (every chapter shift is preceded by 400/800 stop bit signals?)

Gamnn, 2019-12-16:
Appending the chapter number to the ld-decode progress would be useful along with this.

Gamnn, 2019-12-16:
For seeking to side B, scanning for a drop in amplitude would be a useful hint.
lds-amplitude from https://github.com/atsampson/ld-decode-testsuite can do this.

closed 2025-12-09 by simoninns