happycube/ld-decode #60: ld-decode rev3 gets extremely slow as decoding progresses

simoninns, 2018-12-20:
When testing rev3 decoding a 42K frame file, decoding get progressively slower as the decode continues.  Falling to over 1 second per field by about frame 40K.

The issue looks like it's related to the way the JSON output generates a temporary file; the file size (of the main JSON output) is around 16Mb by frame 40K.  ld-decode seems to write the whole thing as a .tmp and then copy it to the main .tbc.json causing a massive amount of file IO operations.

happycube, 2018-12-20:
partial fix (leave this open) - after 100 fields it now writes every 1000.

happycube, 2018-12-24:
Going to leave the partial fix in place for rev3.  Just don't interrupt it after decoding x999 frames and it won't be too bad. ;)

simoninns, 2018-12-31:
This tests as fixed now; closing issue

closed 2018-12-31 by simoninns