happycube/ld-decode #62: Field order output from ld-decode incorrect in JSON metadata (rev3)

simoninns, 2018-12-21:
In a decode of the Almanac disc (side 1) the field order becomes reversed at field 44321, here is a snippet of the JSON output (complete output file attached):

        {
            "isEven": true,
            "syncConf": 75,
            "seqNo": 44319,
            "medianBurstIRE": 14.675160294117648,
            "fieldPhaseID": 3
        },
        {
            "isEven": false,
            "syncConf": 75,
            "seqNo": 44320,
            "medianBurstIRE": 14.630581617647058,
            "fieldPhaseID": 2
        },
        {
            "isEven": false,
            "syncConf": 75,
            "seqNo": 44321,
            "medianBurstIRE": 14.643814705882352,
            "fieldPhaseID": 4
        },
        {
            "isEven": true,
            "syncConf": 75,
            "seqNo": 44322,
            "medianBurstIRE": 14.673281617647058,
            "fieldPhaseID": 3
        },
        {
            "isEven": false,
            "syncConf": 75,
            "seqNo": 44323,
            "medianBurstIRE": 14.676668382352942,
            "fieldPhaseID": 2
        },
        {
            "isEven": true,
            "syncConf": 75,
            "seqNo": 44324,
            "medianBurstIRE": 14.656036397058823,
            "fieldPhaseID": 1
        },

The decode command was as follows:

python3 lddecode.py /media/sdi/Seagate\ Backup\ Plus\ Drive/SFTP/LDV4300D_2\ Captures/Apple\ Visual\ Almanac/Apple\ Visual\ Almanac_CAV_NTSC_side1_2018-11-26_13-12-45.lds /media/sdi/Seagate\ Backup\ Plus\ Drive/AlmanacDecode/almanac_side1_test --length 50000 --start 418

[almanac_side1 JSON.zip](https://github.com/happycube/ld-decode/files/2702805/almanac_side1.JSON.zip)

happycube, 2018-12-21:
per simon this occurs at frame 22160

simoninns, 2018-12-23:
Fixed (rev3)

closed 2018-12-23 by simoninns