happycube/ld-decode #67: ld-analyse:  comb filter button should be sticky...

happycube, 2018-12-23:
... it's kinda annoying having to press it after looking at different frames.

simoninns, 2018-12-23:
I like the idea of this, but the filters are a bit too slow to process for interactive use in the GUI (when perma-on), which is why I implemented it as 'click to filter'.  I'll leave the issue open for now though; might be possible to implement it later.

simoninns, 2018-12-24:
Moved to rev6 as it comes under speed up activities

simoninns, 2018-12-25:
Moved to rev4 'cause @happycube needs it like Sméagol needs his gold ring.

simoninns, 2018-12-25:
This also requires the video extent and drop-out highlighting to function over the comb-filtered frame.

happycube, 2018-12-26:
Thanks for moving this up, it's very nice to have esp with the slider :)

closed 2018-12-26 by happycube