happycube/ld-decode #74: ld-analyse comb filter settings

Gamnn, 2018-12-24:
The ability to set IRE, level, etc. for the comb filter preview would be useful in ld-analyse.

simoninns, 2018-12-24:
The feature is missing from the underlying comb-filter library and needs to be added there first.  I'll look into it.  Primarily adding support for NTSC_J discs with IRE 7.5 is the main issue.

simoninns, 2018-12-24:
This is actually a 'missing' feature in ld-decode.  When processing an NTSC_J disc the output JSON indicates a 7.5IRE black level.  We need to discuss the best way to fix this @happycube

happycube, 2018-12-24:
Done.  It actually was set to NTSC-J, but now defaults to 7.5ire in the front end for NTSC *unless* you use the new --NTSCJ option.

closed 2018-12-24 by happycube