happycube/ld-decode #80: Remove samplesPerUs field from JSON

simoninns, 2018-12-25:
Remove samplesPerUs from the JSON and use the sampleRate instead (as they are redundant)

simoninns, 2018-12-26:
@happycube - the samplesPerUs field has been removed from the tools now and is not longer needed in the JSON output from ld-decode.  Can you remove it from ld-decode and then close this issue?

simoninns, 2018-12-28:
Add blackLevelStart and blackLevelEnd to this list too... They are no longer processed by the tool's shared library and can be removed from the JSON output.  As per commit aa2f818

happycube, 2019-01-02:
Fixed in rev4

simoninns, 2019-01-02:
python3 ld-decode.py ~/lds/Domesday\ Community\ South_CAV_PAL_side2_Disc\ set\ 2_2018-10-02_17-21-20.lds ~/tbc/Domesday_CommS_2 --pal --start 295 --length 60000

Results in:

Namespace(MTF=1.0, MTF_offset=0.0, frame=False, infile='/home/sdi/lds/Domesday Community South_CAV_PAL_side2_Disc set 2_2018-10-02_17-21-20.lds', length=60000, ntsc=False, ntscj=False, outfile='/home/sdi/tbc/Domesday_CommS_2', pal=True, seek=-1, start=295)
Traceback (most recent call last):
  File "ld-decode.py", line 99, in <module>
    write_json(ldd, outname)
  File "ld-decode.py", line 79, in write_json
    jsondict = ldd.build_json(f)
  File "/home/sdi/github/ld-decode/lddecode_core.py", line 1353, in build_json
    spu = vp['samplesPerUs']
KeyError: 'samplesPerUs'

happycube, 2019-01-02:
oops.  fixed.

simoninns, 2019-01-02:
Confirmed

commit 5ce893f3 (2019-01-02): Fix #80
closed 2019-01-02 by simoninns