happycube/ld-decode #84: Using alpha channel for detected dropouts

simoninns, 2018-12-26:
At the moment dropout correction is internal to the ld-decode tool-chain.  It would be good if there was some means to process dropouts using external graphics editing software.  One idea would be to expand the output format (RGB16-16-16) to include an alpha channel; this way detected dropouts could be made 'transparent' as a way to indicate they require some processing in post (an option to ld-dropout-correct perhaps?).

simoninns, 2018-12-26:
Thinking about this a bit; there would need to be an option in ld-dropout-correct to only correct dropouts outside of the active video (visible) picture so that colour burst dropouts are fixed.  Then the comb-filters would need an option to add alpha channel information wherever dropouts occur in the visible area (as non-visible dropout correction requires awareness of the line's phase which cannot be determined in post)

simoninns, 2020-09-16:
Given the overall changes to the ld-decode tool chain and the handling and correction of dropouts in ld-dropout-correct, I no longer see this as a useful enhancement; so I'm closing it.

closed 2020-09-16 by simoninns

Gamnn, 2020-09-16:
I would still find it useful, but as a separate tool that outputs a single-channel grayscale stream rather than using the alpha channel in the rgb.

simoninns, 2020-09-17:
If you want a specialised tool processing a separate stream... simply use the JSON - all of the DOD information is in the metadata.