happycube/ld-decode #90: ld-process-vbi eats .json on full CLV decode

Gamnn, 2018-12-30:
Running ld-process-vbi on a ~80k frame CLV .tbc/.json causes the following to happen:

```
Info: Processing field 168110 16 = "80d000" 17 = "80eeee" 18 = "80eeee"
Info: Processing field 168111 16 = "80d000" 17 = "80eeee" 18 = "80eeee"
Warning: QJson: Document too large to store in data structure (repeated ~8k times)
```
The original .json was about 32MB, and is replaced with a < 1kb file.

simoninns, 2019-01-02:
There seems to be a limit in QJsonDocument of 128Mbytes (which is crazy low IMHO).  So there are two possible fixes... One is to drop the built in JSON library and recode the shared-library for another.  This is a lot of work and will most likely cause a lot of cross-platform issues.  The other is to further optimise the JSON format to bring down the overall file size.

I'm leaning towards the optimisation approach.  The VBI object contains timeCode, statusCode, statusCodeAm2 and clvPicNo sub-objects.  statusCode and statusCodeAm2 are largely redundant copies of each other and, along with the others, could be flattened into the primary VBI object.  For files containing 160,000+ field objects this would represent a significant saving which would (most likely) solve the overall issue of the document size for CLV discs.

simoninns, 2019-01-02:
Proposed new rev4 JSON format for VBI data is now updated in the wiki:

https://github.com/happycube/ld-decode/wiki/JSON-Metadata-format

simoninns, 2019-01-03:
VBI format within JSON has been significantly modified resulting in around a 50% reduction in storage required by fields in the JSON structure.  I've also switched the JSON output format from 'indented' to 'compact' which removes the white-space from the JSON (makes it a little harder to read, but saves a lot of disk space).

Whilst making the fix I've also improved ld-analyse's handling of the VBI so the analysis is now correct for both CAV and CLV discs.

I'm closing this as fixed, but this still needs to be tested with large CLV decodes (which I don't have to hand).  If the problem reoccurs, please open a new issue.

closed 2019-01-03 by simoninns