happycube/ld-decode #96: Audio samples per video field for NTSC and PAL

simoninns, 2019-01-04:
Audio samples per second = 48,000
Video fields per second = 59.9401 (NTSC) or 50.0 (PAL)

Therefore there are 48,000 / 59.9401 = 800.799464799 audio samples per field (NTSC)
or 48,000 / 50 = 960.0 audio samples per field (PAL)

Sample size is 16 bit and each sample is 2 channel; therefore there are 4 bytes per sample

Each NTSC field is  800.799464799 * 4 = 3203.197859196 bytes
Each PAL field is 960 * 4 = 3840 bytes

So, if we take an NTSC and a PAL .tbc containing 400 fields we get:

NTSC 1,281,279.1436784 bytes
PAL 1,536,000 bytes

However, testing ld-decode (on a 400 field decode) gives the following results:

NTSC 1,283,720 bytes
PAL 1,538,460 bytes

So... the question is, what is ld-decode doing and how can I calculate the samples per field rate of the audio .pcm file?  Also, if ld-decode is rounding up/down can you state which way and how?

happycube, 2019-01-06:
Fixed this last night and added audioSamples to field .json.  (audio quality not yet verified ;) )

happycube, 2019-01-06:
It should be 960 for every PAL field and 800 or 801 for NTSC.

simoninns, 2019-01-07:
The integer field audioSamples has also been added to the field object in the JSON to track the number of audio samples associated with the field.

simoninns, 2019-01-07:
When attempting to implement the DOC for audio I noticed I was getting off by one errors for every other field.  Seems ld-decode is varying the field length (based on the first/second field?).

If there is such an association we need to look at the metadata... perhaps there should be a field length for first and second fields rather than just one length for both?

simoninns, 2019-01-08:
Resolved

closed 2019-01-08 by simoninns