happycube/ld-decode #97: Comb filters produce 170Gb+ RGB files

simoninns, 2019-01-04:
The use of RGB16-16-16 as an output format for the comb-filter is generating huge files (bigger than the original disc samples and .tbc files).

There should be options for both the PAL and NTSC comb filters allowing a more compact (as well as more 'standard') RGB output format.

happycube, 2019-01-04:
What I've generally done is pipe comb filter output directly into ffmpeg to produce compressed video.

simoninns, 2019-01-04:
I'm not proposing the removal of RGB16-16-16; but the addition of alternative output formats.

Pipelining is possible to reimplement with the new tool-chain, but not currently supported (as it's less useful for full disc decodes).

happycube, 2019-01-04:
ah... to me, pipelining is *most* useful with full disk encodes.

simoninns, 2019-01-04:
That, as they say, is a whole other issue report :)

simoninns, 2019-05-18:
Both PAL and NTSC filters can now output directly to stdout avoiding the need to create a temporary file

closed 2019-05-18 by simoninns