happycube/ld-decode #100: PAL audio - right channel filter working better than left...

simoninns, 2019-01-07:
When looking at the output from ld-decode (from PAL discs with analogue audio) it's clear that the left channel is far less noisy than the right.  It's possible that the left channel filter is (in some way) not as correct as the right.

[image]

happycube, 2019-06-09:
Hopefully will get a chance to look at it this week, with PAL GGV on it's way to Simon

happycube, 2019-06-28:
Interestingly they pretty much match (albeit with a 2% intensity difference) on ggv1011's 1khz/100% signal.  

What this *really* applies to is audio dropouts, the left channel's are much more intense for some reason.

atsampson, 2019-10-25:
I tried extracting the analogue audio from a badly-rotted copy of BBCL 7044 [using GNU Radio instead](https://github.com/atsampson/ld-decode-testsuite/blob/master/efm/laserdisc.grc) ([screenshot](https://stuff.offog.org/ld/gnuradio/grc.png)), and the "ticks" in the audio tracks are much less severe. Zoom in to see the detail:

[image]

So it looks like this might indeed be a filter artefact...

happycube, 2019-12-05:
This should be greatly reduced in rev6 head.

simoninns, 2019-12-22:
I think this issue can be closed as fixed now - anything else is pretty much covered elsewhere

closed 2019-12-22 by simoninns