happycube/ld-decode #103: Maintaining isFirstField order when dropping frames

simoninns, 2019-01-09:
It's important for the down-stream tools to have a trustworthy isFirstField sequence (otherwise fields can be combined into frames incorrectly).

ld-decode should (if all else fails) create a dummy field when one is dropped and mark the JSON with '-1' in the appropriate fields to show the field is invalid.

simoninns, 2019-01-21:
Partial fix added (now maintains field order, but doesn't correct resulting .tbc video output).  Assigning this issue to @happycube  so he remembers to update it :)

happycube, 2019-01-22:
Moving to rev5 - the partial fix is good for rev4

simoninns, 2019-05-03:
@happycube - Is this issue fixed and just not closed?  Haven't seen the issue pop up for a while :)

happycube, 2019-06-23:
For rev5.2, the previous field of the same type will be repeated.  For rev6 this will be replaced by a totally blank field.

simoninns, 2019-06-23:
repeating a field will also repeat the VBI; care should be taken to test the VBI processes after the 5.2 temporary fix

simoninns, 2020-01-25:
This has been mitigated by the work on ld-discmap which can now spot and correct such issues.  Closing this issue

closed 2020-01-25 by simoninns