happycube/ld-decode #104: Better support for NTSC pull-down

simoninns, 2019-01-09:
The tool-chain needs to provide intelligent support for discs containing NTSC pull-down frames; this is just a general place-holder for the work.

Gamnn, 2019-02-12:
An option to switch to intra-field dropout correction might be a good idea, which going forward would still be good to have on improperly pulled-down field-blended material, which I've found far outnumbers material than can be properly reversed.

simoninns, 2019-05-07:
ld-dropout-correct should now process NTSC pull-down sources correctly with the --intra option.  ld-combine is still 'to be fixed'

simoninns, 2019-06-14:
As a note: A tool should be created to add pull-down field markers into the JSON metadata.  Then the tools could be enhanced to use the metadata when making decisions (especially inter-frame DOC and 3D comb filtering)

simoninns, 2019-08-18:
After a lot of experimentation with ld-discmap; the conclusion is that, the only useful implementation of a pull-down frame detector, would need to use field based image comparison to spot the pull-downs (as there is not reliable way to do this with metadata).  This only affected NTSC CAV discs - NTSC CLV pull-down discs have metadata for the pull-down locations and are now handled by ld-discmap successfully.

Gamnn, 2020-01-09:
Until the toolchain can handle NTSC CAV with unnumbered pull-down frames better, an option in ld-discmap to number all of the non-leadin/leadout frames sequentially would be useful for making the TBCs usable with ld-diffdod/ld-combine.

simoninns, 2020-01-24:
Fixed - Documentation is here: https://github.com/happycube/ld-decode/wiki/Working-with-multiple-discs

closed 2020-01-24 by simoninns