happycube/ld-decode #110: Number of frames produced by --length option incorrect

simoninns, 2019-01-11:
sdi@sehodomedupdev01:~/github/ld-decode$ python3 ld-decode.py /media/sdi/Seagate\ Backup\ Plus\ Drive/SFTP/LDV4300D_2\ Captures/GGV1069/Pioneer\ GGV1069_CAV_NTSC_2018-12-30_22-08-39.lds /media/sdi/Seagate\ Backup\ Plus\ Drive/Rev4_Test/ggv1069_film --start 23764 --length 1799
Namespace(MTF=None, MTF_offset=None, frame=False, infile='/media/sdi/Seagate Backup Plus Drive/SFTP/LDV4300D_2 Captures/GGV1069/Pioneer GGV1069_CAV_NTSC_2018-12-30_22-08-39.lds', length=1799, ntsc=False, ntscj=False, outfile='/media/sdi/Seagate Backup Plus Drive/Rev4_Test/ggv1069_film', pal=False, seek=-1, start=23764)
file frame 23765 CAV frame 23401
file frame 23766 CAV frame 23402
file frame 23767 CAV frame 23403
file frame 23768 CAV frame 23404
file frame 23769 CAV frame 23405
file frame 23770 CAV frame 23406
file frame 23771 CAV frame 23407
file frame 23772 CAV frame 23408
file frame 23773 CAV frame 23409

...

file frame 25558 CAV frame 25194
file frame 25559 CAV frame 25195
file frame 25560 CAV frame 25196
file frame 25561 CAV frame 25197
file frame 25562 CAV frame 25198

file 25562 - file 23765 = 1797
CAV 25198 - CAV 23401 = 1797

Expected result: 1799 frames

happycube, 2019-01-12:
Should be fixed now - it's tracking the actual # of frames written.

simoninns, 2019-01-19:
Closed as fixed

closed 2019-01-19 by simoninns