happycube/ld-decode #112: Decoding failures on Yume kara Semenai

happycube, 2019-01-12:
Gamnn found an issue with an anime disk - Yume kara Samenai (1987) [RLD-1005] similar to the one seen on the National Gallery disk.  This is a tracking bug to remind me to copy in a frame snippet when I collect samples for regression tests :)

three notable failure areas in yumekara2.lds as provided to me:

file frame 101 CLV timecode 2:00.00 frame 3600
WARNING: NTSC field phaseID sequence mismatch
WARNING: NTSC field phaseID sequence mismatch
file frame 102 CLV timecode 2:00.01 frame 3601

file frame 280 CLV timecode 2:05.29 frame 3779
WARNING: NTSC field phaseID sequence mismatch
file frame 281 CLV timecode 2:06.00 frame 3780
WARNING: NTSC field phaseID sequence mismatch
file frame 282 CLV timecode 2:06.01 frame 3781

file frame 434 CLV timecode 2:11.03 frame 3933
WARNING: NTSC field phaseID sequence mismatch
file frame 435 CLV timecode 2:11.04 frame 3934
WARNING: NTSC field phaseID sequence mismatch
file frame 436 CLV timecode 2:11.05 frame 3935

Gamnn, 2019-01-12:
Just a note to check the audio as well, seems buzzy.

happycube, 2019-01-12:
That latest commit fixes it for me.  Sounds nice, in fact...

happycube, 2019-01-15:
Mostly fixed but not completely.  When I fixed it for 1-based fields in lddecode_core.py the pops totally went away, but it caused regressions so it'll probably be a very early rev5 merge.

happycube, 2019-05-19:
Latest code sails through these without framing issues.  Sounds good too :)

commit 0390b2db (2019-02-16): fix #112 - hopefully without more regressions ;)
closed 2019-05-19 by happycube