happycube/ld-decode #118: 3D Comb filter automatic fall back to 2D - NTSC 3D comb filter is broken

simoninns, 2019-01-16:
Whilst trying to fix some related issues around the NTSC comb filter I noticed that the 3D filter wasn't functioning any more; so I rolled back the code a little.

In 3D mode the filter is leaving artefacts in still frames (which should only be processed in 2D).

After a review of the code (both the current and the old non-Qt version) - I can't understand how the code decides when to fallback to 2D.  It looks like the code (when in 3D mode) always applies 3D filtering and the optical flow method is always applied (as there is no decision point based on the 'amount' of flow in a frame).

@happycube - can you take a look at the code and explain to me how it's supposed to work?  Especially the split3D and opticalFlow3D methods in the Comb class (comb.cpp).

Right now the 3D support isn't functioning correctly and I can't fix it without a little more insight into how you designed the original code (especially around how the 3D/2D fallback is supposed to work)

simoninns, 2019-01-16:
It's also very unclear what the noadaptive2d and noopticalflow options are for.

happycube, 2019-01-17:
There's a floating point value for each pixel that determines how much is taken from the 3D CF and the rest from 2D.  This is a typical practice, from what I read...

For noopticalflow - there is/was two 3D comb filter level detectors, one using opencv/optical flow.  That option turned off the OF detector.

simoninns, 2019-01-17:
From IRC:

<simon_dd86> The main thing I need to know on the comb filter is how the 2D/3D fallback works
<simon_dd86> I can't see how the code makes a decision
<happycube> there's a float value that is set to the mix of 2d/3d
<happycube> so in theory where there's high motion, the value should be 0 so the 2d value is taken
<simon_dd86> is there a mix though, or is it one or the other?
<happycube> it's a mix
<simon_dd86> ah, that explains it a bit better then
<happycube> there was an option to see the mix level
<simon_dd86> I shall have a browse through the old code and look for it
<simon_dd86> noadaptive2d and nooptical flow were the other two odd things - what are they for?
<happycube> i forget how i did adaptive 2d, so i think noadaptive2d is just a fallback
<happycube> no optical flow = don't use opencv for motion detection
<simon_dd86> so noopticalflow just turns off the 2D fallback?
<happycube> no
<happycube> it uses a simpler 3D filter
<simon_dd86> ah, ok; which, when I took over the code, didn't seem to ever be called, so I removed it :)
<happycube> then there should be no nopticalflow now then :)
<simon_dd86> hence my confusion about it all I guess

simoninns, 2019-01-17:
Fixed in latest commit

Gamnn, 2019-01-18:
Still seems to be broken in my testing. http://www.framecompare.com/image-compare/screenshotcomparison/YLGLNNNX

simoninns, 2019-01-18:
Looks like the sensitivity of the fall back is too low.  I have no clue how to fix it though; perhaps @happycube can advise?

happycube, 2019-01-18:
Something is weird about these frames - note the reverse 'echos' of frame 2 in frame 1, even in 2D mode.  (If that was just in 3D mode I'd say there was a chroma timing issue where it was using frame 2's data...)

simoninns, 2019-01-18:
Can you suggest some frames from GGV1069 to test it against?  It's probably better if we work from a controlled capture from a known CAV disc with clean source frame separation (captured on a calibrated player with minimal cross-talk).

simoninns, 2019-01-18:
Sounds good; if you give me some frame numbers I can make a capture; I have 2 copies of VE to hand

Gamnn, 2019-01-18:
> Something is weird about these frames - note the reverse 'echos' of frame 2 in frame 1, even in 2D mode. (If that was just in 3D mode I'd say there was a chroma timing issue where it was using frame 2's data...)

Typical field-blended material from poor mastering. Most of my discs are like this unfortunately, so thought it'd be good to test against. the CLD-R7G's 3Dcomb handles it okay.

R7G S-Video capture:
[image]
[image]

I'll test next against one with proper pull-down that I had trouble with though.

simoninns, 2019-01-21:
I've added an optical flow test mode to the comb filter (use --oftest to invoke) - in this mode the filter outputs 3D processed RGB but then superimposes the flow map over the top.  I've done it as a greyscale heatmap (trying to follow (in spirit) what the old filter was doing).

The next issue is that I have no idea what the results actually should look like... @happycube - can you run the filter in oftest mode against a couple of samples and let me know if it matches against your expectations?

Attached is an example of the test mode output.

[image]

simoninns, 2019-01-22:
I've done quite a lot of testing an investigation into the 3D comb filter and I don't see any way it could possibly work... My theory is that (due to the various ways the 3D filter is splitting fields) it hasn't worked since the transition from rev2 to rev3 - and the faults were just pulled into rev4 as-is.

The overall structure of the code makes it very difficult to even understand where to start looking (especially due to the use of global frame buffers which obscure the actual logic and flow).  I think the only practical way forward is to remove the 3D functionality completely, then rewrite the underlying 2D code in a clearer fashion with meaningful variable names and no unnecessary globals.  Once that's done, there is a chance of reintegrating the 3D code in a sane manner.

For rev4 my plan now is to remove the 3D processing and clean up the 2D.  Adding the 3D back in can be pushed to rev5.