happycube/ld-decode #124: libary paths for building on macOS

Gamnn, 2019-01-25:
The homebrew package for macOS installs opencv2 into /usr/local/opt/opencv@2, which needs to be added before the other library paths in ld-analyse.pro and ld-comb-ntsc.pro to build the tools.

+ INCLUDEPATH += "/usr/local/opt/opencv@2/include"
+ LIBS += -L"/usr/local/opt/opencv@2/lib"
   INCLUDEPATH += "/usr/local/include/opencv"
   LIBS += -L"/usr/local/lib"
   LIBS += -lopencv_core -lopencv_imgcodecs -lopencv_highgui -lopencv_imgproc -lopencv_video

simoninns, 2019-03-18:
Closed incorrectly.  Library paths should be added to ld-decode-tools .pro files (provided they do not affect Ubuntu compilation).

simoninns, 2019-03-22:
Fixed in latest rev5 commit

closed 2019-03-22 by simoninns