happycube/ld-decode #126: ld-comb-ntsc 3D artifacts

Gamnn, 2019-01-27:
I've noticed in some black to other color transitions, there are blue and red dot artifacts.

Example:
[image]
[image]
[image]

3rd frame 3dcomb: 
[image]

2dcomb:
[image]

Gamnn, 2019-02-03:
Example 2:
[image]
[image]
[image]

3rd frame 3dcomb:
[image]

2dcomb:
[image]

Gamnn, 2019-02-09:
Example 3:

[image]
[image]
[image]

3rd frame 3dcomb:
[image]

2dcomb:
[image]

happycube, 2020-07-07:
ntsc3d can't pick up the phone right now.

closed 2020-07-07 by happycube