happycube/ld-decode #130: TBC file contained 1 less field than JSON metadata reported

simoninns, 2019-01-28:
Decode command:

python3 ld-decode.py /media/sdi/Seagate\ Backup\ Plus\ Drive/SFTP/LDV4300D_2\ Captures/Apple\ Visual\ Almanac/Apple\ Visual\ Almanac_CAV_NTSC_side1_2018-11-26_13-12-45.lds /media/sdi/Seagate\ Backup\ Plus\ Drive/Rev4_Test/almanac --start 423 --length 850

Expected result 1700 fields in TBC and JSON
Actual result 1699 fields in TBC and 1700 in JSON

happycube, 2019-01-29:
Unable to reproduce here.  .tbc file length was 813722000, which is exactly 1700 * (910 * 263 * 2) bytes.

simoninns, 2019-01-29:
I just retested; and now I get the right result... I have a suspicion that it might have been caused by a mismatch of the ld-decode-tools shared library version on my dev machine :/  I'll close and reopen if I work out what went wrong.

closed 2019-01-29 by simoninns