happycube/ld-decode #132: ld-decode: replace all prints() with writes to stderr

happycube, 2019-01-29:
While looking at ld-cut, I realized it was printing to stdout which makes piping not possible yet.  This is rather silly of me to have done, so it's going into rev5 :)

simoninns, 2019-01-29:
In the C++ apps all output is handled by a single global debug function; which means it's easy to change the way things are displayed to the user and the stream they come out off.  Rather than replacing prints with stderr, why not add a function/class to handle the output and centralize it?  Then you could add things like optional verbose debug and standard info output (like the ld-decode-tools).

I'd be amazed if python doesn't already have some form of output handler; most frameworks do (like Qt's qDebug() class).

happycube, 2019-01-29:
There is, I was just too silly to switch to it earlier...

happycube, 2019-05-19:
Could use refinement, but finally done.

closed 2019-05-19 by happycube