happycube/ld-decode #133: NTSC/3D comb filter: Video Essentials Snell+Wilcox test is mostly processed as 2D CF.

happycube, 2019-01-29:
... it's effectively 2D comb filtering.  The -o option showed lots of movement in static areas.

temporary sample link: https://drive.google.com/open?id=1UYXWhgleD9o0juWL9j5BWiIbaDNFc2Gn

atsampson, 2019-07-04:
I think I might have fixed this accidentally as part of #263.

Here's how the NTSC 3D output looked from rev4 until just before chromamerge:

[image]

And here's how it looks now:

[image]

simoninns, 2019-07-04:
Nice :)  I guess we should run some tests on it.  There is a motion map option which outputs the result of the flow detection I put in there; it's designed so you can see where the code thinks motion is occurring (like a heat-map).

atsampson, 2019-07-04:
So no, I hadn't fixed it, I'd just broken it differently so it was always using the _other_ filter. Here's post-chroma-merge rev6 with `--oftest` (the blob is stationary here):

[image]

Trying different versions, they all have slightly different garbage at the bottom of the frame...

The reason for the garbage is that `OpticalFlow::convertYtoMat` returns a reference to a local -- `cv::Mat` doesn't copy the array you give it. Adding `.clone()` on the end puts it back to how it was in rev4 - it does detect the stationary blob, but it detects spurious movement too:

[image]

atsampson, 2019-08-13:
From a bit of fiddling, a couple of things that seemed to improve the motion detection:

- Subtracting 2 from `velocity` before the `qRound`, so very small movements were ignored.
- Only using one field of the input image. Currently it uses the interlaced frame, so horizontal edges of moving objects will have feathering that changes with the velocity.

I wonder if a much simpler motion detection approach would work better, though...

happycube, 2020-07-07:
ntsc3d doesn't currently exist.

closed 2020-07-07 by happycube