happycube/ld-decode #143: Support for raw file output from HackTV and ignore functions of .json files

Zcooger, 2019-02-05:
Hello, I have some raw composite video samples in PAL format that are using different precision. It will be useful to add support for these formats. They were generated by HackTV transmitter for HackRF:
https://github.com/fsphil/hacktv/blob/c4294769b2948825622be9b78a4d2c6b1bd61bb9/file.c
I provide them here as follows (change extension to raw):
* UINT8: [uint8.raw.txt](https://github.com/Tee0125/yuvplayer/files/2829527/uint8.raw.txt)
* INT8: [int8.raw.txt](https://github.com/Tee0125/yuvplayer/files/2829540/int8.raw.txt)
* INT16: [int16.raw.txt](https://github.com/Tee0125/yuvplayer/files/2829544/int16.raw.txt)
* INT32: [int32.raw.txt](https://github.com/Tee0125/yuvplayer/files/2829551/int32.raw.txt)
* Float: [float.raw.txt](https://github.com/Tee0125/yuvplayer/files/2829563/float.raw.txt)
The resolution is 896 x 625 so the sample rate is 14MHz:
[image]
HackTV files are similar to baseband .tbc files generated from .lds capture bbut may use different precision/bit depth. As shown above HackTV produces split view of both fields at once.
If original .tbc file is replaced by .raw file from HackTV the result is:
[image]
After adjusting output sample rate (17734472 Hz) based on .json file the result is:
[image]

simoninns, 2019-02-05:
The int16 raw format is already supported.  The tools use JSON in order to understand the input format.  Rather than modifying the tools, the simple approach would be to write a JSON generator for the int16 raw format and retest.  The project is very focused on LaserDiscs right now (and there are only two active developers), so it will not be a priority to support HackRF in the near-term.  You are welcome to prepare the changes and submit a pull-request though if you'd like to contribute the code.

Zcooger, 2019-02-06:
So it seems that HackRF should generate JSON files for it's output with fixed values such as IREs, medianBurstIRE, no dropouts etc.
Some extra testing has been done and I think:
- HackTV has messed INT16 output and needs fixing it:
[image]
- HackTV produces accurate output but can produce only signed INT16 big endian (while LD Decode read unsigned INT16 little endian dumps?).
- Endianness does matter:
[image]

[image]
[image]
[image]
[image]

Zcooger, 2019-02-06:
That strong burst may be caused by wrong interpretation of raw dump. As I said this is how Signed is read by LD Decode. Below I show how it should look like (Original HackTV UINT8 output):
[image]
(Levels Corrected HackTV UINT8 output):
[image]
What do you think about these?

simoninns, 2019-02-06:
The .tbc file is uint16 (sorry I mistakenly said int16 earlier) - so the uint8 hack TV (with levels corrected) looks right; you could probably scale this to uint16?  Although the PAL comb filter has been enhanced (by me) to process with 16-bit accuracy that gives you a much nicer final result (especially with images that have wide gradients of colour). 

There is another field in the JSON "medianBurstIRE" that is used to tell the comb-filter what the expected IRE level of the input colour burst is.

Using the oscilloscope view in ld-analyse will also show you where the white and black IRE levels are, so you can see if your video file has the levels correct.

Zcooger, 2019-02-06:
Thanks, so it means HackTV needs some minor changes to fix the sync level and let the user to choose uint16 - the level corrected image was made by simply editing the bitmap in Gimp. Thanks for the explanation. Additionaly in https://github.com/fsphil/hacktv/issues/19 I told the HackTV creator about VITS lines so in the future calibration will be possible. This combo may become the best software CVBS reference generator. I wonder what exact IRE levels will be for HackTV output since the values in JSON come from LDS demodulation, right?

simoninns, 2019-02-06:
The correct IRE levels are according to the PAL specifications (I believe this is ITU-R BT.470-6) but the various signal levels are widely documented on the web.  For example, the IRE width of the colour burst is 40 IRE.  The IRE levels themselves are relative based on the sample format of the thing that generates the output (such as ld-decode or hackTV).

On the subject of VITS; I believe the standards for PAL also specify the minimum VITS types that should be included and on what field lines they should occur.

Zcooger, 2019-02-06:
My commit introducing uint16 to HackTV makes it more compatible with LD Decode, however sync level is by default too high using all data types for CVBS modes and I'll find why.
[image]
[image]
Some chroma troubleshooting will be needed too regardless of JSON modification.

closed 2019-02-07 by Zcooger