happycube/ld-decode #145: PAL Comb filter produces negative color if TBC file has separate fields

Zcooger, 2019-02-07:
[image]
[image]
[image]
[image]

PALDec software can process separate fields at once, but if they are combined negative colors appear. LDDecode can process color properly only on combined fields which normally never happens (unless it is a progressive 626 signal). Any chances for small fix? I have no idea how to correct this.

simoninns, 2019-02-07:
I can't really understand what it is you are raising the issue about...  The comb filter is designed to accept field input, one field at a time and then produce frames by combining two fields.  In your report what do you mean by 'combined fields'?

Zcooger, 2019-02-07:
LDS demodulation program produces a TBC file, that TBC file is been shown above in raw pixel viewer as 576 line frame.
Ok. I used old revision sample as you said. I was not aware of that honestly.

closed 2019-02-07 by Zcooger