happycube/ld-decode #149: Steps to build in Arch (and Manjaro)

toynbeeidea, 2019-02-13:
Arch (and Manjaro) come with opencv4.  opencv2 is available via the AUR, but will not work because Debian or Ubuntu backported some features from opencv3 (or 4) into their opencv2 packages.  Makes no sense to me, but whatever.  The point is, ld-analyse.pro and ld-comb-ntsc.pro need to be edited as follows:

+    INCLUDEPATH += "/usr/include/opencv4" 
+    LIBS += -L"/usr/lib"
  
    INCLUDEPATH += "/usr/local/include/opencv"
    LIBS += -L"/usr/local/lib"
    LIBS += -lopencv_core -lopencv_imgcodecs -lopencv_highgui -lopencv_imgproc -lopencv_video

Arch has different packages than Ubuntu, of course, and there is no separation of -dev packages, and they handle their own Python modules, so pip is not necessary.  The following packages need to be installed:

clang
opencv
fann (AUR)
tk
python-numpy
python-pandas
python-scipy
python-matplotlib
qt5-charts

There might be other packages required that I already had for other purposes, I couldn't say.  This will, at very least, let you compile and launch the software - still need to test it on an actual LD image.

simoninns, 2019-03-18:
Right now the target environment is Ubuntu 18.04 LTS and we'll try to keep the wiki updated with the necessary build steps.  Once the decoder is more stable it may well be worth looking into multi-platform package management but, right now, there simply isn't the dev bandwidth for it.  Closing as won't fix.

closed 2019-03-18 by simoninns

toynbeeidea, 2019-05-21:
> Right now the target environment is Ubuntu 18.04 LTS and we'll try to keep the wiki updated with the necessary build steps. Once the decoder is more stable it may well be worth looking into [...]

So why did you include the MacOS paths, another OS entirely, but reject this?