happycube/ld-decode #151: I cannot use this tools! (loader not defined)

chezsick, 2019-02-17:
I Each times to execute ld-decode.py return
```
Traceback (most recent call last):
  File "./ld-decode/ld-decode.py", line 68, in <module>
    ldd = LDdecode(filename, outname, loader, frameoutput=foutput, system=system, doDOD = not args.nodod)
NameError: name 'loader' is not defined
```
How to fix this? or fix it
Thanks, Sick

happycube, 2019-02-17:
The source file must have the correct file extension.  Most of the samples you'll see are .lds file type.  I'll fix that now to make it the default.

simoninns, 2019-03-17:
Change issue title to reflect problem and labelled.

chezsick, 2019-03-19:
but, how to read from hacktv?

simoninns, 2019-03-20:
@chezsick - the input file must have a .lds extension at the moment in order to work with ld-decode.  Until this issue is resolved you should be able to simply rename your input file.  I'm not sure this will solve your issue (as I have no idea what hacktv file you're talking about, or it's contents).

simoninns, 2020-01-25:
Closing this issue as it looks like any actual problem with ld-decode was solved in #151

closed 2020-01-25 by simoninns