happycube/ld-decode #156: ld-comb-ntsc should output more toolchain-friendly dimensions

Gamnn, 2019-02-21:
Most video codecs and many tools want dimensions divisible by 4, 8, or 16.

Here's two possible dimensions to crop/pad to. I read that being divisible by 16 is ideal, but it will require considerable padding. Divisible by 8 should be fine too, and would only need two lines of padding.

Mod 8: 760x488 - Crop from line 40, col 132, pad 2 lines at bottom.
[image]

Mod 16: 768x496 - Crop from line 40, col 128, pad 10 lines at bottom.
[image]

Gamnn, 2019-02-21:
For reference, this is how it currently crops. Suggested ones above contain more potential information.

754x486 - Cropped from line 44 col 136, padded 4 lines at the top
[image]

simoninns, 2019-03-22:
I guess this isn't NTSC specific?

Gamnn, 2019-03-22:
The suggested dimensions and line to crop at are NTSC specific of course, but needing to be divisible by ideally 16 or 8 also applies to PAL. I don't have enough PAL material to check against.

simoninns, 2019-05-05:
NTSC output dimension now 760x488, PAL output dimensions are now 928x576

closed 2019-05-05 by simoninns