happycube/ld-decode #158: ld-decode:  burst phase adjustment incorrect

happycube, 2019-02-23:
While working on VITS color phase/SNR analysis, I found it's about 90 degrees off - so the I/Q splitting is a bit off.  The output is correct, of course, so it's pretty minor - just makes the code a bit more confusing than it should be...

happycube, 2019-02-26:
Will probably correct this very early in the rev6 cycle.

happycube, 2019-08-19:
Probably a math error in my analysis code.  The color burst aligner performs only a very small shift, which means things are being handled correctly.

closed 2019-08-19 by happycube

atsampson, 2022-05-20:
@happycube `lddecode/core.py` still contains:

```
        # current burst adjustment as of 2/27/19, update when #158 is fixed!
        badj = -1.4
```

Should that be removed now?