happycube/ld-decode #172: Don't know, bombed out on my first cap and it told me to paste a bug report.

toynbeeidea, 2019-03-14:
I don't know what I'm doing yet, but it says to paste this into a bug report, so here I go.

Command issued:  python3 ld-decode.py To-Y.lds output -s 500 --NTSCJ

Result:

ERROR - please paste the following into a bug report:
current sample:  667334000
arguments:  Namespace(MTF=None, MTF_offset=None, frame=False, ignoreleadout=False, infile='To-Y.lds', length=110000, nodod=False, ntsc=False, ntscj=True, outfile='output', pal=False, seek=-1, start=500)
Exception:  index 764239 is out of bounds for axis 0 with size 763626  Traceback:
  File "ld-decode.py", line 99, in <module>
    f = ldd.readfield()
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 1575, in readfield
    f, offset = self.decodefield()
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 1546, in decodefield
    f = self.FieldClass(self.rf, indata, rawdecode, 0, audio_offset = self.audio_offset)
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 1440, in __init__
    super(FieldNTSC, self).__init__(*args, **kwargs)
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 949, in __init__
    self.linelocs2 = self.refine_linelocs_hsync()
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 773, in refine_linelocs_hsync
    zc = calczc(self.data[0]['demod_05'], ll1, self.rf.iretohz(self.rf.SysParams['vsync_ire'] / 2), reverse=False, _count=400)
  File "/home/erik/NASSync/decodetest/lddutils.py", line 292, in calczc
    if data[start_offset] < target:
Traceback (most recent call last):
  File "ld-decode.py", line 99, in <module>
    f = ldd.readfield()
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 1575, in readfield
    f, offset = self.decodefield()
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 1546, in decodefield
    f = self.FieldClass(self.rf, indata, rawdecode, 0, audio_offset = self.audio_offset)
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 1440, in __init__
    super(FieldNTSC, self).__init__(*args, **kwargs)
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 949, in __init__
    self.linelocs2 = self.refine_linelocs_hsync()
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 773, in refine_linelocs_hsync
    zc = calczc(self.data[0]['demod_05'], ll1, self.rf.iretohz(self.rf.SysParams['vsync_ire'] / 2), reverse=False, _count=400)
  File "/home/erik/NASSync/decodetest/lddutils.py", line 292, in calczc
    if data[start_offset] < target:
IndexError: index 764239 is out of bounds for axis 0 with size 763626

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "ld-decode.py", line 110, in <module>
    write_json(ldd, outname)
  File "ld-decode.py", line 86, in write_json
    jsondict = ldd.build_json(ldd.curfield)
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 1840, in build_json
    vp['isSourcePal'] = True if f.rf.system == 'PAL' else False
AttributeError: 'NoneType' object has no attribute 'rf'

happycube, 2019-03-16:
First, try a higher '-s' value, the disk might not be fully spun up and it could be getting confused.
If you're still having problems, use the ld-cut utility with "-l 5" to save off a small chunk of the failure, post it somewhere, and I'll take a lok at it.

toynbeeidea, 2019-03-16:
Sure.  Right now I'm working with gamsnark, hopefully one of the two of us should have something for you to look at shortly.  I think I set the -s value to like 506, and was successful from there.

toynbeeidea, 2019-03-17:
https://drive.google.com/open?id=15zYhH4i4Lal2NvVhuEJ3i1kf4qsrsq3n

happycube, 2019-03-18:
Thanks - I have the sample and can replicate the failure.  Definitely a case of trying to decode before the disk is spun up, but it shouldn't crash.  No ETA on an actual fix yet, but a good catch!

happycube, 2019-05-31:
No longer crashes at beginning of ggv at least.

closed 2019-05-31 by happycube