happycube/ld-decode #173: Automatic seek to first decodable frame in .lds file

simoninns, 2019-03-17:
Based on the work done for EFM decoding; it's possible to use a PLL style algorithm to measure the frequency of sync pulses and predict where the first decodable frame is within a .lds capture.  Basically you isolate the sync pulse using a simple (inaccurate but fast filter) and pass it though a PLL tuned to the expected frame rate.  Once the PLL provides a reference clock at the correct period, you will have the first decodable frames (the lead-in should provide more than enough for the PLL to lock on to).

Not a super-urgent idea but, give the crashes generated by spin-up frames, might be a fairly straight-forward enhancement to ld-decode.

simoninns, 2019-07-14:
This could work like the lead-out code - hunt for the lead-out and then determine the point where lead-out ends; move back 1 frame (to capture the lead-in) and then begin decoding.

happycube, 2019-12-26:
it's slow, but should work fine.

closed 2019-12-26 by happycube