happycube/ld-decode #175: VITS json doesn't seem to make sense

simoninns, 2019-03-22:
When running ld-decode the following JSON output is created for each field object:

`
"vitsMetrics": {
                "whiteSNR": 36.17622350549715,
                "blackLineF1RFLevel": 8088.895227780862,
                "blackLineF1PreTBCIRE": 1.0209512634398417,
                "blackLineF1PostTBCIRE": 1.0969579489067343,
                "blackLineF1PSNR": 38.69639736431755,
                "blackLineF2RFLevel": 8034.571017739778,
                "blackLineF2PreTBCIRE": 0.9323367929677479,
                "blackLineF2PostTBCIRE": 0.9694488703034807,
                "blackLineF2PSNR": 38.77553083882014
            },
`

Why is there metadata for 2 fields in a single field object?  Which one is for the field object that the VITS metadata is contained in?

If this is really pointing to the first and second fields, then the data should be split between the fields (i.e. drop the F1/F2 and put the metadata in the right place).

happycube, 2019-03-28:
Oooops.  It shouldn't be posting VITS data for the first field, at this time.  Thinking about it I can break it up per-field properly and have the one thing that has to be combined (NTSC line 19 color SNR/phase) done for each field and the previousone...

simoninns, 2019-05-05:
Issue appears to be resolved, but the wiki description is not updated.  Please update wiki and close the issue @happycube

happycube, 2019-05-05:
Updated (and changed pal/ntsc IRE50 lines to a generic grey*)

simoninns, 2019-05-06:
This still isn't correct... copying out vitsMetric lines from the JSON I get:

qreal ntscWhiteFlagSNR;
        
        qreal ntscLine19ColorPhase;
        qreal ntscLine19ColorRawSNR;
        
        qreal greyPSNR;
        qreal greyIRE;
        qreal greyRFLevel;
        
        qreal whiteSNR;
        qreal whiteIRE;
        qreal whiteRFLevel;
        
        qreal ntscLine19Burst70IRE;
        qreal ntscLine19Color3DPhase;
        qreal ntscLine19Color3DRawSNR;
        qreal ntscLine19Burst0IRE;
        
        qreal blackLineRFLevel;
        qreal blackLinePreTBCIRE;
        qreal blackLinePostTBCIRE;
        qreal blackLinePSNR;
        qreal syncLevelPSNR;
        qreal syncRFLevel;
        qreal syncToBlackRFRatio;
        
        qreal syncToWhiteRFRatio;
        qreal blackToWhiteRFRatio;

but there are only 17 on the wiki... seems to be quite random what metrics appear where too (at least I don't see a pattern in the json data)

simoninns, 2019-05-06:
There is also the PAL specific palVITSBurst50Level metric in the json file (so 23 metrics in the JSON vs 17 on the wiki)

closed 2019-05-07 by simoninns