happycube/ld-decode #176: Regression - Crash on CLV timecodes a.29->b.0 in rev5

Gamnn, 2019-03-23:
$ python3 ~/github/ld-decode-rev5/ld-decode.py regression.lds -n --NTSCJ test
file frame 1 CLV timecode 19:14.27 frame 34647
file frame 2 CLV timecode 19:14.28 frame 34648
file frame 3 CLV timecode 19:14.29 frame 34649
file frame 4 unknown
ERROR - please paste the following into a bug report:
current sample:  6157818
arguments:  Namespace(MTF=None, MTF_offset=None, daa=False, efm=False, ignoreleadout=False, infile='regression.lds', length=110000, nodod=False, ntsc=True, ntscj=True, outfile='test', pal=False, seek=-1, start=0)
Exception:  int() argument must be a string, a bytes-like object or a number, not 'NoneType'  Traceback:
  File "~/github/ld-decode-rev5/ld-decode.py", line 102, in <module>
    f = ldd.readfield()
  File "~/github/ld-decode-rev5/lddecode_core.py", line 1690, in readfield
    self.processfield(f)
  File "~/github/ld-decode-rev5/lddecode_core.py", line 1948, in processfield
    fi['clvMinutes'] = int(self.clvMinutes)
$ python3 ~/github/ld-decode-rev4/ld-decode.py regression.lds -n --NTSCJ test
file frame 1 CLV timecode 19:14.27 frame 34647
file frame 2 CLV timecode 19:14.28 frame 34648
file frame 3 CLV timecode 19:14.29 frame 34649
file frame 4 CLV timecode 19:15.00 frame 34650
file frame 5 CLV timecode 19:15.01 frame 34651
file frame 6 CLV timecode 19:15.02 frame 34652
file frame 7 CLV timecode 19:15.03 frame 34653
file frame 8 CLV timecode 19:15.04 frame 34654
file frame 9 CLV timecode 19:15.05 frame 34655
End of file
saving JSON and exiting

[CLV_regression.lds.zip](https://github.com/happycube/ld-decode/files/2999364/CLV_regression.lds.zip)

Gamnn, 2019-03-23:
704d08ba71bc7171b440874a7f177eccd99d18da is the first bad commit
commit 704d08ba71bc7171b440874a7f177eccd99d18da
Author: Chad Page <Chad.Page@gmail.com>
Date:   Mon Feb 18 11:16:25 2019 -0800

    fix #153

:100644 100644 376d2d4fc7851e2cb7598b1e4fe7fc2ef8e95fc3 f2569e31782e94493d96697e1afc21afdb43888e M	lddecode_core.py

happycube, 2019-03-23:
Looks like it's (a) missing the last line, there are usually two copies and the first one might be invalid here, and (b) it should make sure clvMinutes is actually there ;)  Should fix tomorrow evening when I'm back home.

happycube, 2019-03-25:
Made it no longer crash, but the root cause isn't fixed yet (I think one of the data lines is good and not being used)

Gamnn, 2019-03-25:
Broken for me.

```
$ python3 ~/github/ld-decode-rev5/ld-decode.py CLV_regression.lds -n --NTSCJ test
ERROR - please paste the following into a bug report:
current sample:  2153928
arguments:  Namespace(MTF=None, MTF_offset=None, daa=False, efm=False, ignoreleadout=False, infile='CLV_regression.lds', length=110000, nodod=False, ntsc=True, ntscj=True, outfile='test', pal=False, seek=-1, start=0)
Exception:  name 'roundfloat' is not defined  Traceback:
  File "~/github/ld-decode-rev5/ld-decode.py", line 102, in <module>
    f = ldd.readfield()
  File "~/github/ld-decode-rev5/lddecode_core.py", line 1690, in readfield
    self.processfield(f)
  File "~/github/ld-decode-rev5/lddecode_core.py", line 1960, in processfield
    fi['vitsMetrics'] = self.computeMetrics(self.firstfield, f)
  File "~/github/ld-decode-rev5/lddecode_core.py", line 1912, in computeMetrics
    metrics_rounded[k] = roundfloat(metrics[k])
```

happycube, 2019-03-25:
ooops, that was an unrelated issue because i added a function to lddutils.py.

Gamnn, 2019-03-25:
Working now.

```
$ python3 ~/github/ld-decode-rev5/ld-decode.py CLV_regression.lds -n --NTSCJ test
file frame 1 CLV timecode 19:14.27 frame 34647
file frame 2 CLV timecode 19:14.28 frame 34648
file frame 3 CLV timecode 19:14.29 frame 34649
file frame 4 unknown
file frame 5 CLV timecode 19:15.01 frame 34651
file frame 6 CLV timecode 19:15.02 frame 34652
file frame 7 CLV timecode 19:15.03 frame 34653
file frame 8 CLV timecode 19:15.04 frame 34654
file frame 9 CLV timecode 19:15.05 frame 34655
End of file
saving JSON and exiting
```

Trouble frame:

[image]
[image]

closed 2019-04-22 by happycube