happycube/ld-decode #177: Pasting a crash (v4)

toynbeeidea, 2019-03-26:
Will provide actual LD sample soon.

file frame 719 CLV timecode 0:23.17 frame 707
file frame 720 CLV timecode 0:23.18 frame 708
ERROR - please paste the following into a bug report:
current sample:  963719149
arguments:  Namespace(MTF=None, MTF_offset=None, frame=False, ignoreleadout=False, infile='tomoe-01.lds', length=110000, nodod=False, ntsc=False, ntscj=True, outfile='tomoe-01', pal=False, seek=-1, start=0)
Exception:  %d format: a number is required, not NoneType  Traceback:
  File "ld-decode.py", line 99, in <module>
    f = ldd.readfield()
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 1593, in readfield
    self.processfield(f)
  File "/home/erik/NASSync/decodetest/lddecode_core.py", line 1788, in processfield
    print("file frame %d CLV timecode %d:%.2d.%.2d frame %d" % (rawloc, self.clvMinutes, self.clvSeconds, self.clvFrameNum, self.frameNumber))

toynbeeidea, 2019-03-26:
From LaserDiscAnalyser:

SampleDetails::getInputSampleDetails(): Sample file is "/home/erik/NASSync/decodetest/tomoe-01.lds" containing 108917686272 samples and is 136147107840 bytes
AnalyseTestData::run(): Thread running
AnalyseTestData::analyseSampleStart(): Reading in 10-bit format
AnalyseTestData::analyseSampleStart(): startSeconds = 0 endSeconds = 2722 total seconds = 2722
AnalyseTestData::analyseSampleStart(): startSample = 0 endSample = 108917686272 samples to analyse = 108917686272
AnalyseTestData::analyseDataIntegrity(): Bad data! Expecting 610 got 470 instead at pointer 1
FileConverter::run(): Processed 102400000 of 108917686272 ( 0 %)

toynbeeidea, 2019-03-26:
https://drive.google.com/file/d/1RtUyLyxnsehJuGBkrSSAMBRS_onQZMSa/view?usp=sharing

Gamnn, 2019-03-26:
Just a note: AnalyseTestData in dddutil (formerly named LaserDiscAnalyser) is only used for test data generated by the DdD when you select that option in the DomesdayDuplicator capture app, not for disc captures.

Can confirm crash in rev4 on file frame 62 of the supplied sample, but seems to be working in the rev5 development branch.

Gamnn, 2019-03-26:
fixed in rev5 by commit cea0fb650ffb46579d83a4339202c4483ac84633.

happycube, 2019-03-28:
worked around in rev4, closing

closed 2019-03-28 by happycube