happycube/ld-decode #179: "early-CLV" crash

toynbeeidea, 2019-04-09:
```
file frame 19460 CLV timecode 10:48.14 frame 19454
file frame 19461 CLV timecode 10:48.15 frame 19455
file frame 19462 early-CLV minute 10
ERROR - please paste the following into a bug report:
current sample:  25976651520
arguments:  Namespace(MTF=None, MTF_offset=None, frame=False, ignoreleadout=False, infile='tomoe-02.lds', length=110000, nodod=False, ntsc=False, ntscj=True, outfile='tomoe-02', pal=False, seek=-1, start=0)
Exception:  int() argument must be a string, a bytes-like object or a number, not 'NoneType'  Traceback:
  File "ld-decode.py", line 99, in <module>
    f = ldd.readfield()
  File "/home/erik/NASSync/ld-decode.4.8.19/lddecode_core.py", line 1593, in readfield
    self.processfield(f)
  File "/home/erik/NASSync/ld-decode.4.8.19/lddecode_core.py", line 1809, in processfield
    fi['frameNumber'] = int(self.frameNumber)

```
Will provide a clip in the next day or so.

toynbeeidea, 2019-04-09:
Hm... I clipped out the part where the crash occurred, only this time it processed the frames in question without any issue, so I'm not quite sure what to do.  Its the same disc that #177 was about, if that helps any.

happycube, 2019-04-22:
Odd, there is a guard around what was 1809, that shouldn't have happened, but it seems to have then... *shrug*  Could be an edge condition of some sort.

happycube, 2019-05-05:
These should be (finally!) fixed now.  This printout area is now protected by a try/except block.

closed 2019-05-05 by happycube