happycube/ld-decode #182: NTSC field phaseID sequence mismatch breaks EFM

Gamnn, 2019-04-22:
Getting "WARNING: NTSC field phaseID sequence mismatch" causes the EFM to break. Starting the decode after the mismatch works fine.

Sample lds: https://drive.google.com/open?id=1-sh1m4XThLjqGNNmstQkgDZ1KOcWa71X

```
file frame 1 CLV timecode 10:55.01 frame 19651
file frame 2 CLV timecode 10:55.02 frame 19652
file frame 3 CLV timecode 10:55.03 frame 19653
file frame 4 CLV timecode 10:55.04 frame 19654
file frame 5 CLV timecode 10:55.05 frame 19655
file frame 6 CLV timecode 10:55.06 frame 19656
file frame 7 CLV timecode 10:55.07 frame 19657
file frame 8 CLV timecode 10:55.08 frame 19658
file frame 9 CLV timecode 10:55.09 frame 19659
file frame 10 CLV timecode 10:55.10 frame 19660
WARNING: NTSC field phaseID sequence mismatch
WARNING: NTSC field phaseID sequence mismatch
file frame 12 CLV timecode 10:55.12 frame 19662
file frame 13 CLV timecode 10:55.13 frame 19663
file frame 14 CLV timecode 10:55.14 frame 19664
file frame 15 CLV timecode 10:55.15 frame 19665
file frame 16 CLV timecode 10:55.16 frame 19666
file frame 17 CLV timecode 10:55.17 frame 19667
file frame 18 CLV timecode 10:55.18 frame 19668
file frame 19 CLV timecode 10:55.19 frame 19669
file frame 20 CLV timecode 10:55.20 frame 19670
```

Gamnn, 2019-04-22:
Frame 11 from the sample:

[image]

happycube, 2019-04-25:
Setting this to rev6.  The root cause on the ld-decode end is a dropout at the beginning of HSYNC in an equalization pulse and that's throwing off the TBC and it really shouldn't.

On the tools end, the EFM processor should recover from these skips.

Gamnn, 2019-04-26:
Here's an interesting one. [lds sample 150MB](https://drive.google.com/open?id=1p8vNUFrznRdYIJAqH2hEyTOor35I571y)

This clip has a series of long dropouts:
[image]

ld-decode causes roughly 3 seconds of garble from the primary problem frame (5).
ldstoefm plays the problem frame fine, but glitches ~2sec after, possibly caused by different dropout. (4 invalid C2s)
ld-decode starting after the problem frame has no glitch.

[image]

problem frame (5): 
[image]

ld-decode:
```
Info: EFM processing:
Info:   Total number of sections processed = 25
Info:   Q Mode 0 sections = 0 (Data)
Info:   Q Mode 1 sections = 0 (CD Audio)
Info:   Q Mode 2 sections = 0 (Disc ID)
Info:   Q Mode 3 sections = 0 (Track ID)
Info:   Q Mode 4 sections = 20 (Non-CD Audio)
Info:   Sections with failed Q CRC = 5
Info: 
Info: EFM to F3 Frame converter:
Info:   Total number of F3 Frames = 21946
Info:   of which 12632 were 588 bits and 9314 were invalid lengths
Info:   Lost frame sync 0 times
Info: 
Info: C1 Level error correction:
Info:   Total number of C1s processed = 21945
Info:   of which 3045 passed and 18900 failed
Info:   The C1 error correction recovered 986 corrupt C1s
Info:   The delay buffer was flushed 0 times
Info: C2 Level error correction:
Info:   Total number of C2s processed = 21837
Info:   of which 2241 passed and 19596 failed
Info:   The C2 error correction recovered 253 corrupt C2s
Info:   The delay buffer was flushed 0 times
Info: C2 Deinterleave:
Info:   Total number of C2s processed = 21943
Info:   of which 2187 were valid and 19756 were invalid
Info:   The delay buffer was flushed 0 times
Info: 
Info: F2 to F1 converter:
Info:   Total number of F1 frames = 0
Info:   Number of F1 frames with sync missing = 0
Info:   Lost F1 sync 0 times
Info: 
Info: F2 Frames to audio converter:
Info:   Total number of stereo audio samples = 21943
Info: 
Info: F3 to section converter:
Info:   Total number of sections = 25
Info:   Number of sections with SYNC0 or SYNC1 missing = 5
Info:   Lost section sync 1 times
```

ldstoefm:
```
Info: EFM processing:
Info:   Total number of sections processed = 219
Info:   Q Mode 0 sections = 0 (Data)
Info:   Q Mode 1 sections = 0 (CD Audio)
Info:   Q Mode 2 sections = 0 (Disc ID)
Info:   Q Mode 3 sections = 0 (Track ID)
Info:   Q Mode 4 sections = 211 (Non-CD Audio)
Info:   Sections with failed Q CRC = 8
Info: 
Info: EFM to F3 Frame converter:
Info:   Total number of F3 Frames = 21579
Info:   of which 21538 were 588 bits and 41 were invalid lengths
Info:   Lost frame sync 0 times
Info: 
Info: C1 Level error correction:
Info:   Total number of C1s processed = 21578
Info:   of which 21414 passed and 164 failed
Info:   The C1 error correction recovered 2020 corrupt C1s
Info:   The delay buffer was flushed 0 times
Info: C2 Level error correction:
Info:   Total number of C2s processed = 21470
Info:   of which 21468 passed and 2 failed
Info:   The C2 error correction recovered 1006 corrupt C2s
Info:   The delay buffer was flushed 0 times
Info: C2 Deinterleave:
Info:   Total number of C2s processed = 21576
Info:   of which 21572 were valid and 4 were invalid
Info:   The delay buffer was flushed 0 times
Info: 
Info: F2 to F1 converter:
Info:   Total number of F1 frames = 0
Info:   Number of F1 frames with sync missing = 0
Info:   Lost F1 sync 0 times
Info: 
Info: F2 Frames to audio converter:
Info:   Total number of stereo audio samples = 21576
Info: 
Info: F3 to section converter:
Info:   Total number of sections = 219
Info:   Number of sections with SYNC0 or SYNC1 missing = 8
Info:   Lost section sync 0 times
```

ld-decode skipping problem frame:
```
Info: EFM processing:
Info:   Total number of sections processed = 211
Info:   Q Mode 0 sections = 0 (Data)
Info:   Q Mode 1 sections = 0 (CD Audio)
Info:   Q Mode 2 sections = 0 (Disc ID)
Info:   Q Mode 3 sections = 0 (Track ID)
Info:   Q Mode 4 sections = 205 (Non-CD Audio)
Info:   Sections with failed Q CRC = 6
Info: 
Info: EFM to F3 Frame converter:
Info:   Total number of F3 Frames = 20720
Info:   of which 20692 were 588 bits and 28 were invalid lengths
Info:   Lost frame sync 0 times
Info: 
Info: C1 Level error correction:
Info:   Total number of C1s processed = 20719
Info:   of which 20579 passed and 140 failed
Info:   The C1 error correction recovered 1569 corrupt C1s
Info:   The delay buffer was flushed 0 times
Info: C2 Level error correction:
Info:   Total number of C2s processed = 20611
Info:   of which 20611 passed and 0 failed
Info:   The C2 error correction recovered 812 corrupt C2s
Info:   The delay buffer was flushed 0 times
Info: C2 Deinterleave:
Info:   Total number of C2s processed = 20717
Info:   of which 20717 were valid and 0 were invalid
Info:   The delay buffer was flushed 0 times
Info: 
Info: F2 to F1 converter:
Info:   Total number of F1 frames = 0
Info:   Number of F1 frames with sync missing = 0
Info:   Lost F1 sync 0 times
Info: 
Info: F2 Frames to audio converter:
Info:   Total number of stereo audio samples = 20717
Info: 
Info: F3 to section converter:
Info:   Total number of sections = 211
Info:   Number of sections with SYNC0 or SYNC1 missing = 7
Info:   Lost section sync 0 times
```

simoninns, 2019-05-03:
I'm not convinced this is really an issue with the tools (the EFM decode will work if passed valid EFM), so I'm un-assigning myself from this issue.

happycube, 2019-05-05:
Should be fixed now

closed 2019-05-05 by happycube