happycube/ld-decode #183: Crash (on rotter): Cannot convert float NaN to integer

Gamnn, 2019-04-24:
```
$ ~/github/ld-decode-rev5/ld-decode.py
[issue183.lds.zip](https://github.com/happycube/ld-decode/files/3111748/issue183.lds.zip)
 issue183.lds issue183 -n --NTSCJ
file frame 1 CAV frame 573
file frame 2 CAV frame 574
file frame 3 CAV frame 575
~/anaconda3/lib/python3.7/site-packages/numpy/core/fromnumeric.py:2920: RuntimeWarning: Mean of empty slice.
  out=out, **kwargs)
~/anaconda3/lib/python3.7/site-packages/numpy/core/_methods.py:85: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
WARNING: TBC failure at line 1
WARNING: TBC failure at line 2
WARNING: TBC failure at line 3
WARNING: TBC failure at line 4
WARNING: TBC failure at line 5
WARNING: TBC failure at line 6
WARNING: TBC failure at line 7
WARNING: TBC failure at line 8
WARNING: TBC failure at line 9
WARNING: TBC failure at line 10
WARNING: TBC failure at line 11
WARNING: TBC failure at line 12
WARNING: TBC failure at line 13
WARNING: TBC failure at line 14
WARNING: TBC failure at line 15
WARNING: TBC failure at line 16
WARNING: TBC failure at line 17
WARNING: TBC failure at line 18
WARNING: TBC failure at line 19
WARNING: TBC failure at line 20
WARNING: TBC failure at line 21
WARNING: TBC failure at line 22
WARNING: TBC failure at line 23
WARNING: TBC failure at line 24
WARNING: TBC failure at line 25
WARNING: TBC failure at line 26
WARNING: TBC failure at line 27
WARNING: TBC failure at line 28
WARNING: TBC failure at line 29
WARNING: TBC failure at line 30
WARNING: TBC failure at line 31
WARNING: TBC failure at line 32
WARNING: TBC failure at line 33
WARNING: TBC failure at line 34
WARNING: TBC failure at line 35
WARNING: TBC failure at line 36
WARNING: TBC failure at line 37
WARNING: TBC failure at line 38
WARNING: TBC failure at line 39
WARNING: TBC failure at line 40
WARNING: TBC failure at line 41
WARNING: TBC failure at line 42
WARNING: TBC failure at line 43
WARNING: TBC failure at line 44
WARNING: TBC failure at line 45
WARNING: TBC failure at line 46
WARNING: TBC failure at line 47
WARNING: TBC failure at line 48
WARNING: TBC failure at line 49
WARNING: TBC failure at line 50
WARNING: TBC failure at line 51
WARNING: TBC failure at line 52
WARNING: TBC failure at line 53
WARNING: TBC failure at line 54
WARNING: TBC failure at line 55
WARNING: TBC failure at line 56
WARNING: TBC failure at line 57
WARNING: TBC failure at line 58
WARNING: TBC failure at line 59
[... 199 lines cut ...]
WARNING: TBC failure at line 259
WARNING: TBC failure at line 260
WARNING: TBC failure at line 261
WARNING: TBC failure at line 262
WARNING: TBC failure at line 263
ERROR - please paste the following into a bug report:
current sample:  4988693
arguments:  Namespace(MTF=None, MTF_offset=None, daa=False, efm=False, ignoreleadout=False, infile='/Volumes/Cappy/RF Capture/Majo demo Steady/decode/issue183.lds', length=110000, nodod=False, ntsc=True, ntscj=True, outfile='issue183', pal=False, seek=-1, start=0)
Exception:  cannot convert float NaN to integer  Traceback:
  File "~/github/ld-decode-rev5/ld-decode.py", line 102, in <module>
    f = ldd.readfield()
  File "~/github/ld-decode-rev5/lddecode_core.py", line 1734, in readfield
    f, offset = self.decodefield()
  File "~/github/ld-decode-rev5/lddecode_core.py", line 1705, in decodefield
    f = self.FieldClass(self.rf, indata, rawdecode, 0, audio_offset = self.audio_offset)
  File "~/github/ld-decode-rev5/lddecode_core.py", line 1467, in __init__
    self.downscale(wow = True, final=True)
  File "~/github/ld-decode-rev5/lddecode_core.py", line 1422, in downscale
    dsout, dsaudio, dsefm = super(FieldNTSC, self).downscale(audio = final, *args, **kwargs)
  File "~/github/ld-decode-rev5/lddecode_core.py", line 937, in downscale
    self.dsaudio, self.audio_next_offset = downscale_audio(self.data[1], lineinfo, self.rf, self.linecount, self.audio_offset)
  File "~/github/ld-decode-rev5/lddecode_core.py", line 598, in downscale_audio
    output_left = np.mean(audio['audio_left'][np.int(locs[i]):np.int(locs[i+1])])

```

Gamnn, 2019-04-24:
Here's another spot that triggers it:
[issue183_2.lds.zip](https://github.com/happycube/ld-decode/files/3112823/issue183_2.lds.zip)

happycube, 2019-04-25:
Hopefully the TBC improvements I have in mind for #184 will help here too...

happycube, 2019-05-05:
Should be fixed now

commit 2a28b3d9 (2019-05-01): reduce sensitivity of an error detector to fix #183
closed 2019-05-05 by happycube