happycube/ld-decode #185: Segfault in ld-dropout-correct

Gamnn, 2019-04-29:
I'm getting a segfault with this TBC file on Ubuntu 18.04.

$ ld-dropout-correct sample_3.tbc sample3_doc.tbc
Info: Frame # 1 [ 2 / 3 ] - 33 dropouts corrected
Info: Frame # 2 [ 4 / 5 ] - 195 dropouts corrected
Info: Frame # 3 [ 6 / 7 ] - 44 dropouts corrected
Info: Frame # 4 [ 8 / 9 ] - 28 dropouts corrected
Info: Frame # 5 [ 10 / 11 ] - 175 dropouts corrected
Info: Frame # 6 [ 12 / 13 ] - 23 dropouts corrected
Info: Frame # 7 [ 14 / 15 ] - 135 dropouts corrected
Info: Frame # 8 [ 16 / 17 ] - 53 dropouts corrected
Info: Frame # 9 [ 18 / 19 ] - 22 dropouts corrected
Segmentation fault (core dumped)

Works on macOS 10.12.6 most of the time, but sometimes crashes in the same spot with" Bus error: 10" instead of the segmentation fault message.

I'm unable to reproduce this with smaller samples, so here's the 150 frame trouble .tbc (which I could reproduce):
https://drive.google.com/open?id=1Wo00yW7IPX-wD1lW-EcgWRkh6uA5WfGh

happycube, 2019-04-30:
It reproduces for me, and Valgrind shows a lot of errors in different places.  issues/185/issue185-valgrind.log in ld-decode-testdata has a dump.

simoninns, 2019-05-03:
Fixed in latest commit.  Please retest and confirm.

closed 2019-05-05 by simoninns