happycube/ld-decode #186: Regressions in EFM

Gamnn, 2019-05-01:
Two in one here:

every decode starts with a 'blip' in the right channel, and a section of a .lds which decoded to digital pcm previously is now broken.

[image]

[issue186.lds](https://drive.google.com/open?id=1WHDv7isjXkYbiDfNubFf0MQEKjE1EiSY)

happycube, 2019-05-01:
(mostly a note to self to remind me what I still need to go over later, since I went over it on IRC with Gamnn)

The new framing code added more lines so it was badly corrupting the .efm output.  I fixed that by using the beginnings of VSYNC, but that's vulnerable to glitching so I need to be able to resolve the beginning of VSYNC from the known good pulse locations in those cases.

I don't have any idea what's going on with the beginning glitches yet, I saw it on a direct read from he010 as well so it's definitely a thing.

simoninns, 2019-05-05:
The channel glitch in the EFM audio decode was being caused by a bug in ld-process-efm.  Basically it wasn't waiting for F2 frames before outputting data (EFM needs 109 frames to fill the interleave buffers before any valid audio is produced).  This issue is fixed in the latest commit (so it may be that you can close this issue, but I'm unsure about the rest of the discussion above, so I will leave it open for now).

happycube, 2019-05-07:
I think this is fixed now.  (re-open if not.)

commit a0ff00f2 (2019-05-01): mostly fix #186
closed 2019-05-07 by happycube