happycube/ld-decode #188: DOC artifact

Gamnn, 2019-05-04:
The DOC seems to be getting the phase wrong on this dropout, causing an ugly artifact.

[image]
[image]
[image]
[image]

[docglitch.lds.zip](https://github.com/happycube/ld-decode/files/3144255/docglitch.lds.zip)

Gamnn, 2019-05-04:
Possibly related to https://github.com/happycube/ld-decode/issues/104?

simoninns, 2019-05-06:
Is the sample provided an NTSC disc with pull-down @Gamnn ?  If so, this isn't a bug - the DoC doesn't yet work correctly with pull-down discs (which is covered by #104) since the field order is moving (so correcting non-pulldown frames from pull-down frames causes phasing issues).

Gamnn, 2019-05-06:
It wasn't properly mastered, so not proper pull-down, and the frame in question doesn't appear interlaced.
I didn't look too closely, but I wondered if it might still be the fault of the intrafield DOC that #104 might have a solution for.

simoninns, 2019-05-06:
ok, I'm going to close this issue as it's probably already covered by #104 which will have to include an option to force intrafield DoC rather than the inter-field DoC used for non-pull down sources

closed 2019-05-06 by simoninns